simonsobs/sotodlib #672: HWP freq should be in obsdb for SATs

mhasself, 2024-01-24:
Mean and stdev are written into the M_index.yaml:
```
/so/data/satp1/obs/17058/obs_1705877708_satp1_1111111/M_index.yaml:hwp_freq_mean: 2.043986363358419
/so/data/satp1/obs/17058/obs_1705877708_satp1_1111111/M_index.yaml:hwp_freq_stdev: 0.0017446569674796405
/so/data/satp1/obs/17058/obs_1705878069_satp1_1111111/M_index.yaml:hwp_freq_mean: 2.042640613243446
/so/data/satp1/obs/17058/obs_1705878069_satp1_1111111/M_index.yaml:hwp_freq_stdev: 0.0005945323034167648
```

Some obs have nulls:
```
/so/data/satp1/obs/17053/obs_1705340592_satp1_1111111/M_index.yaml:hwp_freq_mean: null
/so/data/satp1/obs/17053/obs_1705340592_satp1_1111111/M_index.yaml:hwp_freq_stdev: null
```

So we'll have to decide what to do with those.  Maybe null is ok.

kmharrington, 2024-01-24:
@AERAdler and I were just chatting today about what we needed to add these to existing obsdbs

mhasself, 2025-06-24:
This is resolved (with the caveat that direction is not in yet, but that's on its way).

closed 2025-06-24 by mhasself