simonsobs/sotodlib #681: tod_ops.flags.get_turnaround_flags should produce a RangesMatrix with [dets, samps] shape

chervias, 2024-01-27:
The function for creating turnaround flags creates the RangesMatrix with axis samps, but maybe it should be axes [dets, samps]. For the filterbin mapmaking, the turnaround flags will go directly into the cuts argument of the pointing matrix operation, so it would be useful to have it ready to go considering it usually will be OR with other flags, and it should have the correct shape.

kmharrington, 2024-02-07:
oh, instead of making it a `RangesMatrix` with dets,samps we should make it a Ranges of length samps. `Ranges` * `RangeMatrix` logic is supported iirc

kmharrington, 2024-02-14:
I just found where this might be showing up. Has to do with where multiplication is defined for RangesMatrix. `RangesMatrix * Ranges` works as you'd like it to. `Ranges * RangesMatrix` does not. Luckily flags are commutative

msilvafe, 2024-02-15:
So @kmharrington it sounds like you prefer we keep the turnaround flags as Ranges instead of padding into a RangesMatrix? If so I think we can close this issue.

chervias, 2024-02-20:
I looked into this and I simply need to do `RangesMatrix` + `Ranges` to get the behavior I want. This is because I keep thinking of them as matrices when they are more than that. I'll close the issue.

closed 2024-02-20 by chervias