simonsobs/sotodlib #684: TypeError in sotodlib.toast.ops.sim_sso ?

ykyohei, 2024-01-30:
I find following error.
Tested in site computing system, kernel SO shared (py310)
Used latest sotodlib today (8b3b1b71a17d1f676290449227e5b5136882a797).
Is this because of my environment?

```
from sotodlib.toast.ops import sim_sso

...

TypeError: concatenate() got an unexpected keyword argument 'dtype'
```

tskisner, 2024-01-30:
Not sure if it is related, but just a heads-up that I just (an hour ago) tagged a toast release and wheels are updated.  There was a trivial import change introduced in that which is fixed in #683, which we are trying to get merged.  However, if you are using the previous toast tag (i.e. not from today) and current sotodlib master, then this should not be the problem.

mhasself, 2024-01-30:
> TypeError: concatenate() got an unexpected keyword argument 'dtype'

You should show the error traceback.  Some people can read them!

But if that's a failure in numpy.concatenate, the dtype arg should be present from 1.20 onwards.  The shared software is 1.25:
```
>>> import numpy
>>> numpy.__file__
'/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/numpy/__init__.py'
>>> numpy.__version__
'1.25.0'
```

mhasself, 2024-01-30:
This might have been a transient error while the toast package was being updated in the shared software system...

You might find it is magically working now...

ykyohei, 2024-01-30:
I still get this error so I copy and paste the traceback.
```
(so-base-py310) ykyohei@compute22:~$ python3
Python 3.10.12 (main, Jul  5 2023, 18:54:27) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import numpy
>>> numpy.__version__
'1.25.0'
>>> from sotodlib.toast.ops import sim_sso
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/so/shared-software/so-unstable-py310/lib/python3.10/site-packages/sotodlib/toast/ops/__init__.py", line 9, in <module>
    from .act_sign import ActSign
  File "/so/shared-software/so-unstable-py310/lib/python3.10/site-packages/sotodlib/toast/ops/act_sign.py", line 10, in <module>
    from toast.ops import Operator
  File "/so/shared-software/so-unstable-py310/lib/python3.10/site-packages/toast/ops/__init__.py", line 11, in <module>
    from .conviqt import SimConviqt, SimTEBConviqt, SimWeightedConviqt
  File "/so/shared-software/so-unstable-py310/lib/python3.10/site-packages/toast/ops/conviqt.py", line 9, in <module>
    import healpy as hp
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/healpy/__init__.py", line 95, in <module>
    from .rotator import Rotator, vec2dir, dir2vec
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/healpy/rotator.py", line 43, in <module>
    .transform_to("galactic")
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/coordinates/sky_coordinate.py", line 675, in transform_to
    new_coord = trans(self.frame, generic_frame)
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/coordinates/transformations.py", line 1479, in __call__
    curr_coord = t(curr_coord, curr_toframe)
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/coordinates/transformations.py", line 1252, in __call__
    newrep = self._apply_transform(fromcoord, *params)
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/coordinates/transformations.py", line 1174, in _apply_transform
    rep = rep.transform(matrix)
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/coordinates/representation.py", line 1399, in transform
    p = erfa_ufunc.rxp(matrix, self.get_xyz(xyz_axis=-1))
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/coordinates/representation.py", line 1351, in get_xyz
    return np.stack([self._x, self._y, self._z], axis=xyz_axis)
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/units/quantity.py", line 1683, in __array_function__
    return super().__array_function__(function, types, args, kwargs)
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/numpy/core/shape_base.py", line 456, in stack
    return _nx.concatenate(expanded_arrays, axis=axis, out=out,
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/units/quantity.py", line 1688, in __array_function__
    args, kwargs, unit, out = function_helper(*args, **kwargs)
TypeError: concatenate() got an unexpected keyword argument 'dtype'
>>>
```

ykyohei, 2024-01-30:
Yeah, I got same error from import healpy, so this is not an issue of sotodlib but an issue of shared env?
```
(so-base-py310) ykyohei@compute22:~$ python3
Python 3.10.12 (main, Jul  5 2023, 18:54:27) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import astropy
>>> import healpy
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/healpy/__init__.py", line 95, in <module>
    from .rotator import Rotator, vec2dir, dir2vec
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/healpy/rotator.py", line 43, in <module>
    .transform_to("galactic")
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/coordinates/sky_coordinate.py", line 675, in transform_to
    new_coord = trans(self.frame, generic_frame)
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/coordinates/transformations.py", line 1479, in __call__
    curr_coord = t(curr_coord, curr_toframe)
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/coordinates/transformations.py", line 1252, in __call__
    newrep = self._apply_transform(fromcoord, *params)
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/coordinates/transformations.py", line 1174, in _apply_transform
    rep = rep.transform(matrix)
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/coordinates/representation.py", line 1399, in transform
    p = erfa_ufunc.rxp(matrix, self.get_xyz(xyz_axis=-1))
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/coordinates/representation.py", line 1351, in get_xyz
    return np.stack([self._x, self._y, self._z], axis=xyz_axis)
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/units/quantity.py", line 1683, in __array_function__
    return super().__array_function__(function, types, args, kwargs)
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/numpy/core/shape_base.py", line 456, in stack
    return _nx.concatenate(expanded_arrays, axis=axis, out=out,
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/astropy/units/quantity.py", line 1688, in __array_function__
    args, kwargs, unit, out = function_helper(*args, **kwargs)
TypeError: concatenate() got an unexpected keyword argument 'dtype'
```

mhasself, 2024-01-31:
Ya, this seems to be a version incompatibility between the astropy and numpy we have installed.  Looks like [it was addressed](https://github.com/astropy/astropy/blob/b259f3f1978e565776049ac64cc2445f78a11c37/CHANGES.rst#L1519) in astropy 5.1.1 (we were running 5.1) -- so I'm upgrading it on the shared stack... or trying to... will get back to you.

mhasself, 2024-01-31:
Ok, astropy is now at 5.3.4, and all the imports you mentioned seem to work!

closed 2024-01-31 by mhasself