simonsobs/sotodlib #689: context.get_meta can interrupt site_pipeline?

ykyohei, 2024-02-03:
I knew that keep opening the metadata h5 file interrupts `site_pipeline`, but I was assuming `get_meta` does not.
However, I observed that the `get_meta` sometimes interrupts `make_hwp_solution.py` during the process of re-producing the `hwp_angles`,

This happened while I was watching the log of `make_hwp_solution.py`  in real time, so I'm relatively confident that something other than `get_meta` did not cause the interruption.

I did following in the jupyter-lab.
```
ctx = core.Context("/so/home/ykyohei/scratch/ctx_satp3.yaml")
aman = ctx.get_meta('obs_1704138534_satp3_1111111')
```

Following is the log of `make_hwp_solution.py`.
The error looked synchronized with the execution of the `get_meta`.
```
10412.115: Calculating Angles for obs_1704161020_satp3_1111111 (INFO)
WARNING: sotodlib.core.context: Key "metadata" not present in context.
WARNING: sotodlib.hwp.g3thwp: flipping quad is corrected by mean value
ERROR: sotodlib.hwp.g3thwp: Exception '[Errno 11] Unable to open file (unable to lock file, errno = 11, error message = 'Resource temporarily unavailable')' thrown while the template subtraction.
WARNING: sotodlib.hwp.g3thwp: Packet drop exists
WARNING: sotodlib.hwp.g3thwp: 1 dropped packets are found, performing fill process
WARNING: sotodlib.hwp.g3thwp: Offcentering calculation is only available when two encoders are operating. Skipped.
Traceback (most recent call last):
  File "/so/home/ykyohei/tools/sotodlib/sotodlib/hwp/g3thwp.py", line 994, in write_solution_h5
    aman.save(output, h5_address, overwrite=True)
  File "/so/home/ykyohei/tools/sotodlib/sotodlib/core/axisman.py", line 871, in save
    return _save_axisman(self, dest, group=group, overwrite=overwrite)
  File "/so/home/ykyohei/tools/sotodlib/sotodlib/core/axisman_io.py", line 191, in _save_axisman
    file_to_close = h5py.File(dest, 'a')
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/h5py/_hl/files.py", line 567, in __init__
    fid = make_fid(name, mode, userblock_size, fapl, fcpl, swmr=swmr)
  File "/so/shared-software/miniconda3/envs/so-base-py310/lib/python3.10/site-packages/h5py/_hl/files.py", line 243, in make_fid
    fid = h5f.open(name, h5f.ACC_RDWR, fapl=fapl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5f.pyx", line 106, in h5py.h5f.open
BlockingIOError: [Errno 11] Unable to open file (unable to lock file, errno = 11, error message = 'Resource temporarily unavailable')

10679.998: Calculating Angles for obs_1704167870_satp3_1111111 (INFO)
WARNING: sotodlib.core.context: Key "metadata" not present in context.
WARNING: sotodlib.hwp.g3thwp: Packet drop exists
WARNING: sotodlib.hwp.g3thwp: 1 dropped packets are found, performing fill process
WARNING: sotodlib.hwp.g3thwp: Packet drop exists
WARNING: sotodlib.hwp.g3thwp: 2 dropped packets are found, performing fill process
```

kmharrington, 2024-02-06:
yes, this is a "feature" of using hdf5 files as the mechanism by which we save data and running through an automated setup. My suggestion is for these two improvements to be included as part of the make-hwp-solutions development: 

(1) the hwp angle hdf5 files are massive. we need a mechanism for automatically splitting into new files after some amount of time or data. See this issue for some vague words on directions of implementation: https://github.com/simonsobs/sotodlib/issues/670 

(2) `make-hwp-angles` should be made robust to temporary external locks. Ex: try except the IO errors, wait 5 seconds, try again (if number of trys < some value ~5).

ykyohei, 2024-02-23:
(1) I think the simplest way to split h5 file is to output different h5 file for each obs_id. Is this okay? Do we have a reason for needing a mechanism to automatically split into new files after some amount of time or data?

mhasself, 2024-02-23:
> (1) I think the simplest way to split h5 file is to output different h5 file for each obs_id. Is this okay? Do we have a reason for needing a mechanism to automatically split into new files after some amount of time or data?

Lots of small files is annoying.  You at least have to organize them so you don't end up with 1000s of files in a single directory.  But then you find yourself doing a lot of mkdir in addition to writing out the files.

Can I request that you take the obs_id, grab the timestamp out of it, and then take the first 4 digits?  That file will hold about 2 weeks of data.  E.g.  `angles_1670.h5`  How big will that file get?  If it's over, I dunno, 50 GB then we should probably do 5 digits instead of 4.

It would be nice if site_pipeline.util provided some help with this, but it doesn't really yet.  But generating filenames like that ^^^ is pretty easy, and solves the lots-of-files problem.

ykyohei, 2024-02-23:
Okay, that sounds good and easy enough for me to implement. Thank you for your suggestion.

closed 2024-03-06 by ykyohei