simonsobs/sotodlib #693: Preprocess plotting

davidvng, 2024-02-06:
I'm working on integrating plotting into each step of preprocess and wanted to get comments from sotodlib folks (I've talked with @msilvafe about this but particularly wanted opinions from @kmharrington ). The idea is to add the option to automatically produce plots after each preprocess step, which would then be served to Teleview (or some other web service). The types of plots would depend on the step, specifically separating steps that have `calc` (`calc_and_save`) vs those that have `process`.

- After each `calc_and_save` step, make plots specific to that step's calculation (I have an example [here](https://github.com/simonsobs/sotodlib/compare/master...preprocess-plots) for `get_det_bias_flags`)
- After each `process` step, make an updated PSD plot. Then maybe make one at the end with all of them overplotted like in [this draft PR](https://github.com/simonsobs/sotodlib/pull/522).

This plotting would be an option to set to True in the preprocess config. The path and naming conventions are other topics of discussion. I was just going to follow the typical 5 digit timecode structure along with the `obs_id` and the plot type (ex: `/17047/obs_1704770693_satp1_1111111_bias_cuts.png`)

kmharrington, 2024-02-06:
I think the biggest first thing to decide on is "inside the tod-ops functions" or "inside the preprocess modules." For your first example, you aren't actually doing anything with the preprocess modules so it doesn't really have to do with the `calc_and_save` step.  That's really editing the plotting into the tod-ops functions directly to include a plot/save option.

I think I'm leaning toward we should do this all as an expansion to the preprocess modules and not inside the functions directly. That would continue to separate the "functions that do things to/with the TODs" from the "results/statistics from those functions." Basically it keeps all the IO out of `tod-ops`. ( TBH lots of overlap between plotting and generating statistics so there are a bunch of thoughts in this PR: https://github.com/simonsobs/sotodlib/pull/691 ) I think I would do this by editing the modules `calc_and_save` or `save` functions to look for specific plotting keywords and then building plots there. The `calc_and_save` function is responsible for calling the `save` function so you can be flexible about what goes between those two. And `calc_and_save` is called after `process` so it will catch changes made to there. 

The one downside to "inside preprocess modules" is it's not totally clear to me what the overhead looks like for somehow who "just wants to run this function and plot what things look like." To it's worth thinking if there's an easy way to make things flexible enough for that to be easy. 

The second idea you're discussing, where we plot a PSD (or something) after each step. I think that either takes something like Tomoki did (config file with a whole bunch of PSDs and a wrap up function). Or it involves making the preprocess pipeline more than just the list it currently is. TBH you could convince me of either depending on how things are structures / setup.

Other implementation comments / notes:
* [Importing pyplot before setting a matplotlib backend](https://github.com/simonsobs/sotodlib/blob/d446646cc6fd9911fbf9fa26f810af4b168cf8e7/sotodlib/tod_ops/flags.py#L5) breaks things when running on clusters / remote machines without display set. (I know pysmurf does it, it's bad). @mhasself  do we have a standard for matplotlib in sotodlib? If not yet then I'd expect at a minimum we need to have `matplotlib.use('agg')` before any pyplot import statement.
* [We should not run an AxisManager restrict inside a flag generation function, definitely not inside a plot if statement.](https://github.com/simonsobs/sotodlib/blob/d446646cc6fd9911fbf9fa26f810af4b168cf8e7/sotodlib/tod_ops/flags.py#L92).

msilvafe, 2024-02-07:
Hey @kmharrington just to give you my thinking on this as I set @davidvng in this direction then asked him to post this discussion before he got too far...so just based off of how I have been generating plots to understand data quality I thought a natural divide would be for all preprocess `calc` steps we basically generate some sort of calculation (flags, fitted HWPSS, fitted az polynomials, white noise, etc.) these require some custom plot to visualize that particular type of calculation which I thought best fit in the base function in `tod_ops` and then could be called in the preprocess config with a boolean argument like `make_plot`. For any of the preprocess `process` steps that actually modify the detector signal somehow we could have a standardized function like `record_qa` discussed in PR: https://github.com/simonsobs/sotodlib/pull/691 which just plots the TOD + PSD. It could be more complicated and collect all such TODs + PSDs and make one plot at the end but I'm not really set on that, I think plots might become hard to read and it requires more infrastructure (as you mentioned in reference to Tomoki's draft PR https://github.com/simonsobs/sotodlib/pull/522)

closed 2024-10-01 by msilvafe