simonsobs/sotodlib #701: Two Pi Jumps missing what seems like obvious 2pi jumps

kmharrington, 2024-02-13:
I am testing on the `jump-heights` branch but I think the code I'm running is in master. I found some examples where it really seems like we're missing some 2pi jumps but I don't understand the function enough to understand why we're missing them. 

```
obs_id = 'obs_1698801938_satp1_1101111'
dets = ['sch_ufm_mv7_1698194790_6_136', 'sch_ufm_mv7_1698194790_6_196']
ctx = core.Context("/so/metadata/satp1/contexts/basic.yaml")
aman = ctx.get_obs( obs_id=obs_id, dets=dets)

jumps, heights = tod_ops.jumps.twopi_jumps(aman, merge=False, fix=False)
print(jumps.get_stats())
plt.figure()
for i in range(0, aman.dets.count):
    plt.plot(aman.timestamps, aman.signal[i])
```

Gets us no found jumps but here's the plot: 
[image]

@skhrg  could you take a look? thanks!

skhrg, 2024-02-14:
Will take a look, but may have to punt for ~1 week until the I get back from the site.

kmharrington, 2024-02-15:
I now have a whole database of missed jumps 
```
missed_database = {
    'obs_1698441314_satp1_1111111' : [
        'sch_ufm_mv9_1698194187_0_170', 
        'sch_ufm_mv9_1698194439_2_423',
        'sch_ufm_mv9_1698194837_5_143', 
        'sch_ufm_mv9_1698194978_6_219'
    ],
    'obs_1698457111_satp1_1111111' :[
        'sch_ufm_mv19_1698196257_0_445',
        'sch_ufm_mv19_1698196695_3_387',
        'sch_ufm_mv19_1698196995_6_226',
        'sch_ufm_mv19_1698196995_6_248',
        'sch_ufm_mv19_1698196995_6_306',
        'sch_ufm_mv19_1698196995_6_412',
        'sch_ufm_mv7_1698194470_4_223',
        'sch_ufm_mv7_1698194470_4_471',
        'sch_ufm_mv7_1698194970_7_106',
        'sch_ufm_mv7_1698194970_7_431',
        'sch_ufm_mv7_1698194618_5_211'
    ],
    'obs_1698525273_satp1_1111111':[
        'sch_ufm_mv19_1698196257_0_459',
        'sch_ufm_mv19_1698196400_1_170',
        'sch_ufm_mv19_1698196400_1_387',
        'sch_ufm_mv19_1698196400_1_504',
        'sch_ufm_mv19_1698196551_2_160',
        'sch_ufm_mv19_1698196551_2_170',
        'sch_ufm_mv19_1698196551_2_262',
        'sch_ufm_mv7_1698194970_7_434',
        'sch_ufm_mv7_1698194470_4_060',
        'sch_ufm_mv7_1698194790_6_098',
        'sch_ufm_mv7_1698194790_6_316',
    ],
    'obs_1698785421_satp1_1111111':[
        'sch_ufm_mv19_1698196257_0_360',
        'sch_ufm_mv19_1698196551_2_134',
        'sch_ufm_mv19_1698196551_2_136'
        'sch_ufm_mv7_1698194618_5_511',
    ],
    'obs_1698801938_satp1_1101111':[
        'sch_ufm_mv19_1698196995_6_114',
        'sch_ufm_mv19_1698196400_1_410',
        'sch_ufm_mv19_1698196400_1_423',
        'sch_ufm_mv19_1698196551_2_027',
        'sch_ufm_mv19_1698196551_2_326',
        'sch_ufm_mv19_1698196695_3_445',
        'sch_ufm_mv19_1698196995_6_002',
        'sch_ufm_mv19_1698196995_6_026',
        'sch_ufm_mv19_1698196995_6_159',
        'sch_ufm_mv19_1698196995_6_178',
        'sch_ufm_mv19_1698196995_6_179',
        'sch_ufm_mv19_1698196995_6_248',
        'sch_ufm_mv19_1698196995_6_258',
        'sch_ufm_mv19_1698196995_6_267',
        'sch_ufm_mv19_1698196995_6_396',
        'sch_ufm_mv19_1698196995_6_432',
        'sch_ufm_mv19_1698197109_7_099',
        'sch_ufm_mv7_1698194618_5_024',
        'sch_ufm_mv7_1698194790_6_136',
        'sch_ufm_mv7_1698194790_6_196',
        'sch_ufm_mv7_1698194970_7_207',
        'sch_ufm_mv7_1698194970_7_215',
    ]
}
```

closed 2024-03-04 by skhrg