simonsobs/sotodlib #712: Terminate Preprocess Pipeline when n_dets = 0

msilvafe, 2024-02-18:
When running a preprocess pipeline and the select method restricts so that the axis manager has no detectors left the next step in the pipeline will fail. This happens occasionally with the detector bias cuts due to tuning in very bad weather or failed tuning. We need to catch this case and make the pipeline exit gracefully (i.e. appropriately update databases with a no_data, or no_detectors all steps after the select) so that we can batch run on all observations with the `preprocess_tod.py` site-pipeline script.

kmharrington, 2024-02-20:
Yes. I suggest implementing some sort of results dictionary to `pipeline.run()` for status things. So that we can do something like

```
proc_aman, status = pipe.run(aman)
if not status['success']:
    do something about status['message'] 
    ... etc
```

msilvafe, 2024-03-21:
Addressed in #751