simonsobs/sotodlib #713: Missing preprocess modules

msilvafe, 2024-02-18:
We need some additional preprocess modules/updates:

1. PCA - calculating the PCA, could add a select based off of this later but want to use it now to calculate a relcal.
2. Fourier filtering - I want to be able to pass a lowpass version of the data to the PCA, and as a last step in the pipeline apply a highpass filter to kill 1/f.
3. Expand Calibrate method - This method currently just multiplies all detectors in signal by the same calibration factor. To apply a relcal or abscal we will want to be able to calibrate w/ a vector with a different factor per detector.

kmharrington, 2024-02-20:
For (3) I was thinking we could add a method that just pulls a len(dets) array from a field in an axismanger. where most times it will be `det_cal&phase_to_pW`. But we could also end up stacking those modules where we do that one and then also a rel-cal and then also and abs-cal. But "pull field from axismanager" will work for all of those.

BrianJKoopman, 2024-03-05:
A solution for 3 is implemented in #743.

closed 2024-10-01 by msilvafe