simonsobs/sotodlib #722: Have det_bias_flags work on meta as well as aman

kmharrington, 2024-02-21:
Right now the creation of the mask fails on a loaded meta object because it doesn't have a samps axis. It'd be nice for this cut to able able to run before we load all the data and then cut all the detectors. Having to load the whole tod first is a waste of memory. 

Example of what I would like to do
```
meta = ctx.get_meta(obs)
flag = tod_ops.flags.get_det_bias_flags(meta, new_key_arg?)
meta.restrict( 'dets', meta.vals.dets[ flag ])

aman=ctx.get_aman(meta)
```

msilvafe, 2024-02-21:
I can take this issue. Is it sufficient to check if there’s a samps axis and if so return a ranges matrix and if not return a Boolean mask?

msilvafe, 2024-03-21:
Fixed in PR #734

closed 2024-03-21 by msilvafe