simonsobs/sotodlib #763: Raise error in making projection object when NaN is in tod.focal_plane

tterasaki, 2024-03-13:
I (we) have wasted a lot in debugging when forgetting restricting detectors with non-NaN pointing result.

skhrg, 2024-03-13:
Spent a hot second debugging the same thing with @tdalford  the other day.

Worth noting that `aman.det_info.wafer.type == "OPTC"` should get you a mask that removes all the nan pointing (since only non optical dets should have them if you are using the full focal plane). So maybe something that tries to apply that by default would be useful?

msilvafe, 2024-03-13:
We had discussed having the focal plane manifestdB load in the flags you want to use to restrict to just the optical detectors in addition to the focalplane axis manager as is can we do that and then default to having the mapping functions use that set of flags?

skhrg, 2024-03-13:
Yeah I think that makes sense. To be clear though do we want this flag to make its way into `aman.flags` or just live in `aman.focal_plane`?

msilvafe, 2024-03-13:
I think `aman.focal_plane` is fine for now. Eventually it would be good to collect all flags into `aman.flags` (there are many in `aman.preprocess`) but I feel like maybe we want that to be a modification of the loader that checks for flags in the metadata and loads them into flags.

msilvafe, 2024-03-14:
I also like that solution. Maybe more.

mhasself, 2024-03-21:
That solution is a great mitigation for now (maybe test gamma instead, though ... sometimes gamma is the only nan, but that's enough to segfault).

I could imagine it being convenient to leave fp.* as is, and suppress unpointed dets using det_weights.  But that is a bit more work to add in.

davidvng, 2024-04-17:
Hi @mhasself , I'm interested in your suggestion of suppressing unpointed dets using det_weights. Could you give more details on what you're thinking? I've ran into this issue when trying to run https://github.com/simonsobs/sotodlib/blob/cb49a57d587f9bfed19a13026cd73df816f3eae5/sotodlib/coords/planets.py#L425 which eventually calls https://github.com/simonsobs/sotodlib/blob/cb49a57d587f9bfed19a13026cd73df816f3eae5/sotodlib/coords/helpers.py#L283 which fails to converge due to the NaNs

mhasself, 2024-04-18:
> which fails to converge due to the NaNs

Because there can be NaNs in focal plane, any method that "does stuff" with focal_plane.{xi,eta,gamma} has to be robust against that.  Either by raising exception, or just omitting those dets when appropriate.

I believe the convergence failure for get_footprint is actually from [get_focal_plane_cover](https://github.com/simonsobs/sotodlib/blob/cb49a57d587f9bfed19a13026cd73df816f3eae5/sotodlib/coords/helpers.py#L381).  It can easily be made "NaN" safe.  That is not really related to the "projection" problem, which was (prior to [recent so3g fixes](https://github.com/simonsobs/so3g/releases/tag/v0.1.13)) straight up segfaulting.  Someone should go through coords.* and just screen out places where focal_plane is used in a NaN-unsafe way!

closed 2024-04-26 by mhasself