simonsobs/sotodlib #776: Include sotodlib version in hwp_angle metadata

ykyohei, 2024-03-21:
Adding `sotodlib.__version__` to the metadata is helpful because metadata identifiers do not necessarily reflect the version of the sotodlib, and there may be minor inconsistencies in an hwp metadata.
For example, debugging may only be applied to observations after a certain time.

mhasself, 2024-04-10:
> Whatever soconda does ... you don't get a version code. I don't know why that is.

Oh --- this is probably because of a shallow clone (depth=1 or whatever).  That would be missing the tag information required to construct the version code.

tskisner, 2024-04-10:
Right, currently we are just installing sotodlib master, but I think what I will do going forward (unless there are objections) are to make an sotodlib tag whenever we are going to update the stack and then use that tag for the install.  Would that have any side-effects?

mhasself, 2024-04-11:
> Right, currently we are just installing sotodlib master, but I think what I will do going forward (unless there are objections) are to make an sotodlib tag whenever we are going to update the stack and then use that tag for the install. Would that have any side-effects?

Seems fine to me!

iparask, 2024-04-18:
The tags will also help to quickly reproduce environment variables. I'm in favor of such a side-effect!

closed 2024-05-15 by ykyohei