simonsobs/sotodlib #788: `get_turnaround_flags` needs to handle no az motion

msilvafe, 2024-03-30:
Currently if the Az is stationary `get_turnaround_flags` errors out with an error like this:

```
  File "/so/home/msilvafe/.local/lib/python3.10/site-packages/sotodlib/preprocess/processes.py", line 647, in process
    tod_ops.flags.get_turnaround_flags(aman, **self.process_cfgs)
  File "/so/home/msilvafe/.local/lib/python3.10/site-packages/sotodlib/tod_ops/flags.py", line 199, in get_turnaround_flags
    part_slices_ta_masked = [slice(None, np.where(_ta_flag)[0][0]), slice(np.where(_ta_flag)[0][-1], None)]
IndexError: index 0 is out of bounds for axis 0 with size 0
```

We need to catch this sort of error and return flags that are all false (or appropriately instantiated). A known satp1 obs that fails in this way is: `obs_1706255231_satp1_1111111` and can be used for testing.

mmccrackan, 2024-09-27:
For more recent versions of `get_turnaround_flags` the error output of observations where the Az does not vary significantly is:

`"/global/homes/m/mmccrack/so_home/repos/20240819_catchup/sotodlib/sotodlib/tod_ops/flags.py", line 216, in get_turnaround_flags`
    `peaks, _ = find_peaks(np.abs(matched), height=peak_threshold, distance=rel_distance_peaks*approx_samps_onescan)`
`"/global/u1/m/mmccrack/so_home/env/soconda_20240903_0.1.4/lib/python3.11/site-packages/scipy/signal/_peak_finding.py", line 941, in find_peaks
    raise ValueError('distance must be greater or equal to 1')`

This occurs because `np.ptp` is used to find the largest distance between max and min of Az, but this is small for stationary Az observations and may not meet the requirement `rel_distance_peaks*approx_samps_onescan` be greater than 1.  Some might be a few samples bigger though and would not fail.

msilvafe, 2025-02-19:
I think this is actually ok to fail, if one actually wanted to analyze a stare observation they they wouldn't be calling turnaround flagging.

closed 2025-02-19 by msilvafe