simonsobs/sotodlib #808: Standard TODs for testing

skhrg, 2024-04-15:
Would be really nice if we had a standard set of test TODs (probably one per telescope and both with and without HWP for the SATs) that have typical noise and behavior that can be used to characterize changes to flags.

msilvafe, 2024-10-01:
@kwolz and @earosenberg can I enlist your help with this? I think you've already started on some of this for the glitch + jump investigations.

kwolz, 2024-10-02:
@msilvafe yes - this will largely align with my investigations on false-positive jumps, for which a bunch of TODs need to be studied (and checked for "standard" behavior) anyways.

earosenberg, 2024-10-02:
Hi @msilvafe sure, happy to help out / cross-check with Kevin. Similar story here, I'm looking at lots of TODs to check the glitch finder, and was planning to pick a few from broad categories (e.g. "normal" obs, problematic obs with many glitches, time-varying noise causing current finder to fail) as test obs.
I guess Saianeesh was just looking for the "normal" ones here?

skhrg, 2024-10-02:
I think all of those are valuable. But I also think that "normal" is the most valuable since it lets us test potential improvements with more confience since presumably most TODs will look like the "normal" one.

But long term I 100% agree that a zoo of TODs with examples of less common but expected wierd TODs would be awesome.

kwolz, 2024-10-17:
Pointing to the jump processing validation script [here](https://github.com/simonsobs/pwg-scripts/tree/master/flp/sat-map-processing/kwolz/scripts). The short-term goal (for myself) would be to look at a bunch of SAT obs_ids, run the script and save the summary statistics on detectors (jumps by eye, jumps by auto-detection, false negatives, false positives, unsuccessful fixes, etc.).

Not related to "standard" jumps (but possibly pretty non-standard ones), @chervias  mentioned the drone data is full of false-negative jumps, so it would be good to also cross check with those TODs.

chervias, 2024-10-17:
@kwolz  Once nersc is back I will post here a particularly bad obs, so you can try it if you want.

chervias, 2024-10-19:
The drone flight with the artificial source on is bad. @kwolz I ran your script on `obs_1714507261_satp3_1101111` `ws0` `f090`. I have attached some of the plots produced by the script. When you have time it would be great if you could have a look at this particular tod. thanks!
[image]
[image]
[image]

msilvafe, 2024-10-19:
> The drone flight with the artificial source on is bad. @kwolz I ran your script on `obs_1714507261_satp3_1101111` `ws0` `f090`. I have attached some of the plots produced by the script. When you [...]

Definitely interested in @kwolz investigation. Something to look out for, early on looking at Moon data we were seeing a higher rate of jumps (qualitatively I never quantified this) when we hit the moon (i.e. bright source). I would be interested to know if you're seeing that here with the drone.

kwolz, 2024-10-23:
@chervias @msilvafe I have run my jump validation script on the above SATp3 TOD and have summarised my findings [here](https://simonsobs.atlassian.net/l/cp/S5Mb0SJk). In a nutshell, there are 0 false negatives, 5 false positives, 5 failed fixes. For the latter two categories, the jump finder gets confused by the high-frequency noise/glitchy regions in the TODs and tends to overcorrect.

skhrg, 2024-10-23:
Would be very interested if that is better than @chervias remembers for the drone processing. One of the things I focused on the the rewrite was making it less susceptible to thinking bright sources were jumps but I tested that on very naively injected gaussians.

One thing that can also be done to reduce the false positives is to switch from using `nsigma` to using `atol`. I suspect the issue is that in super noise TODs the `nsigma` based thresh is too lenient which is causing those false positives. Perhaps sticking in an standard typical `atol` is the correct thing to do in those cases. Or maybe I should add a `max_tol` that puts a bound on the `nsigma` derived one.