simonsobs/sotodlib #813: Planet data download from astropy

iparask, 2024-04-18:
Function [`_get_astrometric`](https://github.com/simonsobs/sotodlib/blob/master/sotodlib/coords/planets.py#L239) tries to download planet data. This download does not work on HPC systems like Della and Tiger, where the compute nodes are not connected to the internet. The same will be true for Tiger 3.

In addition, the data are downloaded in the user's home directory by default, which requires every user to have a copy of this data. 

Astropy has a plan to provide configurable paths for caching ([Issue here](https://github.com/astropy/astropy/issues/10738)), but it does not seem to have a lot of traction.

Proposal: `sotodlib` checks first to a specific path, trying to load the necessary data, and falls back in trying to download them if they do not exist while issuing a warning

mhasself, 2024-04-18:
Yes, this has always bothered me... part of the reason for the global "RESOURCE_URLS" variable ... so some process could auto-download and cache everything that might be needed.

An envvar (SOTODLIB_RESOURCES?) for the path, or is there a more modern solution?  The "context.yaml" could also be used (using the [existing logic and envvar defs](https://github.com/simonsobs/sotodlib/blob/cb49a57d587f9bfed19a13026cd73df816f3eae5/sotodlib/core/context.py#L46) to load a "site configuration" file).

mhasself, 2024-04-19:
> @mhasself can I take this task and have two subtasks?

Yes, please go ahead! (1) solves the immediate problem so do feel free to stop there if you want.

PR #817 (2024-07-08): Generalize outside data paths
This PR closes #813.

It introduces an environment variable for data that are downloaded. This variable is a serialized JSON.

This is a breaking change since it will raise a runtime error when the environment variable does not have the expected entry.

We may want to merge it along with an addition to soconda module file that set the variable.
closed 2024-07-08 by iparask