simonsobs/sotodlib #838: preprocess submodule namespace collision

mhasself, 2024-05-10:
`preprocess` submodule directory contains:
```
__init__.py
core.py
preprocess_plot.py
processes.py
```

And that `__init__.py` contains `from processes import *`.  And processes.py imports sotodlib.core as core.  This makes it confusing to access the stuff in sotodlib/preprocess/core.py, because module namespace `sotodlib.preprocess.core` maps back to `sotodlib.core`.

Possible solutions:
- don't `import *`; or if you do then use a restrictive `__all__` declaration in processes.py to only pull in those names you really need (and core is not one of them).
- rename the file preprocess/core.py to something that's not already a name imported into preprocess module.

Here's an example of confusing behavior:
```
>>> from sotodlib.preprocess.core import _expand
>>> _expand
<function _expand at 0x743b7ec3ff70>

>>> from sotodlib.preprocess import core
>>> core._expand
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'sotodlib.core' has no attribute '_expand'
```

msilvafe, 2024-05-10:
I'll work on fixing this.

msilvafe, 2024-05-11:
Addressed in PR #841

closed 2024-05-11 by msilvafe