simonsobs/sotodlib #897: Overhead to read the sparce weight map

tterasaki, 2024-07-03:
I found it takes long to read pixel weight file.
It is just because we are saving weight with shape of (3, 3, ny, nx) although non-diagonal part of 
weight is zero by definition filter-bin mapmaking. 
As shown in the screen shot of test, of course though, it takes 3 times longer to read weight than to read map. 
[image]

Yuji pointed me out that must be painfull when we do null tests. So I suggest that we save the diagonal part of weight in map making and pad the off-diagonal part in using the weight.

chervias, 2024-07-05:
There is a very easy way of doing this with numpy.diagonal, I'm writting it here so I remember later

chervias, 2024-07-17:
PR #905 now has a fix for this.