simonsobs/sotodlib #966: [Critical] a part of the fourier filter method in tod_ops.filters can change all demod signal to nan

hnakata-JP, 2024-09-26:
I found this part in tod_ops.filters.fourier_filter erase or remove all the signals in the demodulation.
Especially, when I called hwp.demod_tod, the demodQ and demodU as wrapped signals came out as nan. I think this can cause critical trouble in map-making.

> logger.info('fourier_filter: IFFT.')
> t_2()

I am not sure of the details, but the problem can be solved temporarily when you comment this out.
**[To experts]** Please check this item, and treat it properly.

hnakata-JP, 2024-09-26:
As an example, you can refind this issue with this obs_id, obs_1725654705_satp1_1111111, but not with the previous one, obs_1725654492_satp1_1111111. This may come out just when you want to check some systematic effect.

hnakata-JP, 2024-09-27:
This issue can occur when the phasor or hwp_angle has a nan value in its array. One can avoid this by restricting TOD to the valid range.

ykyohei, 2024-10-08:
This happens in the observation that the hwp stopped in the middle of observations, so this does not become a problem for the cmb observations with no problem in hwp operation.
We can mitigate this by changing the timestamp interpolation method, and this is addressed in this draft PR https://github.com/simonsobs/sotodlib/pull/931

msilvafe, 2024-10-10:
> This happens in the observation that the hwp stopped in the middle of observations, so this does not become a problem for the cmb observations with no problem in hwp operation.
>
> We can mitigate this [...]

I see, thank you!

ykyohei, 2024-10-11:
I made separate PR that will fix this nan issue, because other hwp angle pipeline update that I plan in https://github.com/simonsobs/sotodlib/pull/931 will take some more time for merging it.

ykyohei, 2024-10-20:
This should be fixed by https://github.com/simonsobs/sotodlib/pull/1000

closed 2024-10-20 by ykyohei