simonsobs/sotodlib #992: New det_cal db introduces NaNs in timeconst_filter

mmccrackan, 2024-10-08:
For a new site pipeline run of satp1 data, I updated the manifest directory to point to a more recent det_cal db (switched from satp1_det_cal_240312m to satp1_det_cal_240910m). It seems that this introduced NaNs into the axis_manager signal during the 
 `timeconst_filter` Fourier filter stage.  Using the older det_cal does not produce this issue.  @jlashner and @msilvafe.

An example observation where this occurs is:
obsid='obs_1717988397_satp1_1111111'
dets={'wafer_slot': 'ws3', 'wafer.bandpass': 'f150'}

The config filter params are:

```
- name: "fourier_filter"
      process:
        filt_function: "timeconst_filter"
        filter_params:
          timeconst: "det_cal.tau_eff"
          invert: True
```

mhasself, 2025-09-05:
@mmccrackan Is this still a concern?

mmccrackan, 2025-09-05:
This should be no longer relevant.  A preprocess function has been added (https://github.com/simonsobs/sotodlib/pull/1249) to restrict on detectors with `NaN` values for `tau_eff` and `phase_to_pW` on both running and loading preprocessing data.