simonsobs/sotodlib #1008: Add support in `LoadContext` for running preprocessing up through demodulation

tskisner, 2024-10-17:
If we are running a preprocessing pipeline during load, and that pipeline config has demodulation enabled, then create demodulated toast observation containers rather than the usual ones.  The mapping between axismanager and toast detector naming conventions should be extracted to helper functions callable by the work needed to address #1007 .

msilvafe, 2024-10-17:
Are you just planning to scan the preprocess config file or loaded pipeline class attributes for this?

msilvafe, 2024-10-17:
I don't think so! You can just look for the config pipeline element with `name == 'demodulate' ` https://github.com/simonsobs/sotodlib/blob/f33fdd8306bcc0c65a64b33f6439ddae8f926db5/sotodlib/preprocess/processes.py#L688