simonsobs/sotodlib #1009: Align processes in Sept 2024 Filter+Bin Mapmaker script with preprocessing module

susannaaz, 2024-10-17:
This issue is to refactor the filter-bin (F+B) mapmaking script starting from the one used for the latest run of SATP3 maps [(make_filterbin_map_20240924.py)](https://github.com/simonsobs/pwg-scripts/blob/master/flp/sat-map-processing/sazzoni/make_filterbin_map_20240924.py) to use the preprocessing module. 

This involves taking away from the (F+B) script all the "on-the-fly" functions that already exist in the module and add to the preprocessing module any new functions: 

- [ ] Rotation to telescope coordinates
- [ ] Subscan poly filtering
- [ ] Demod common mode deprojection
- [ ] AZSS subscan correction 
- [ ] Modified noise fitting (bounds)
- [ ] White noise cuts - low wn, high wn, fknee
- [ ] Apply pointing model --> used for SATP1 

This will also involve:
- [ ] Adding the option to call these functions from the preprocessing module directly inside the F+B script
- [ ] Adding the option to write the outputs of each step into a preprocessing database and/or read directly from a preprocessing database when this exists (this partially overlaps with #973)

msilvafe, 2024-11-13:
Just so you know @susannaaz, I believe that one part of this update is to include subscan level flags based on PSD stats per subscan @earosenberg has opened a draft PR #1028 addressing this as well as adding in the glitch detection criteria (Interquartile range) defined per subscan to reduce the number of false-positives. So you don't need to add that.

msilvafe, 2025-05-05:
Addressed in the merging of the `iso-sat-review` branch #1176

closed 2025-05-05 by msilvafe