simonsobs/sotodlib #1025: preproc_or_load_group in core

chervias, 2024-11-08:
This function currently lives in the site_pipeline section, but probably is better to be in the library side, for example should be in core next to get_obs

msilvafe, 2024-11-08:
Ya, thanks for creating an issue. @mmccrackan, and I have been speaking about the number of functions in site_pipeline that are being called outside of just the main script. I agree we need to move things out, we can discuss where the best place to do that is. We had moved some of the functions into `site_pipeline/util.py` however do you think that it's better that this goes into `preprocess/common.py` or `preprocess/util.py`. @mhasself since you raised this in @chervias's PR review perhaps you have an opinion?

mhasself, 2024-11-08:
Yes, as a general rule, don't import things from site_pipeline into other submodules.  And don't import other submodules into core.

So this functionality should live in preprocess/ submodule.  I think `common` doesn't quite capture the audience for this function so `util` or `main` might be better.  But not a strong feeling on that.

chervias, 2024-11-19:
Hi, is this fixed then by the new PR? If so I'll close the issue

mmccrackan, 2024-11-19:
Yes, this is addressed in #1027.  Closing.

closed 2024-11-19 by mmccrackan