simonsobs/sotodlib #1073: Source flags needs to handle if the planet doesn't cross the scan

msilvafe, 2024-12-19:
The preprocess method [SourceFlags](https://github.com/simonsobs/sotodlib/blob/e5ecdd2a1d9cd3975b1efd4b02da1c502643fecb/sotodlib/preprocess/processes.py#L1013) calls [coords.planets.compute_source_flags](https://github.com/simonsobs/sotodlib/blob/e5ecdd2a1d9cd3975b1efd4b02da1c502643fecb/sotodlib/coords/planets.py#L465) which currently fails if the planet is not present in the observation. We need to run this source flagging step on all CMB observations because we let the moon and other SSOs (Jupiter, Saturn, Mars, Venus, etc.) cross through our CMB field but we don't want them binned into the maps. However the planets are only present in a few observations. So we either need to call `planet.coords.get_scan_q()` in `SourceFlags` to check if the planet is in the observation first before calling `compute_source_flags` or add the logic into `compute_source_flags` after the call to `get_scan_P`.

msilvafe, 2024-12-19:
@RemingtonGerras pointed out [coords.planets.get_nearby_sources](https://github.com/simonsobs/sotodlib/blob/e5ecdd2a1d9cd3975b1efd4b02da1c502643fecb/sotodlib/coords/planets.py#L402) perhaps we can use that to do the logic check in `SourceFlags`

mmccrackan, 2025-01-06:
Addressed in #1079.

closed 2025-01-06 by mmccrackan