simonsobs/sotodlib #1119: obsdb Error: Incorrect Quotes Causing Errors

Ankurdev-astro, 2025-02-20:
I found a bug while instantiating an ObsFileDb object, using the latest build from _master_ branch:
```
from sotodlib.core import metadata
obsfiledb = metadata.ObsFileDb()
```

gives an error:

```
File "/opt/miniforge3/envs/latest-sotodlib/lib/python3.11/site-packages/sotodlib/core/metadata/obsfiledb.py:97, in ObsFileDb.__init__(self, map_file, prefix, init_db, readonly)
     95 self.prefix = self._get_prefix(map_file, prefix)
     96 if init_db and not readonly:
---> 97     self._create()

File "/opt/miniforge3/envs/latest-sotodlib/lib/python3.11/site-packages/sotodlib/core/metadata/obsfiledb.py:182, in ObsFileDb._create(self)
    178     q = ('create table if not exists `%s` (' % table_name  +
    179          ','.join(column_defs) + ')')
    180     c.execute(q)
--> 182 if self._get_version(conn=c) is None:
    183     c.execute('insert or ignore into meta (param,value) values (?,?)',
    184               ('obsfiledb_version', 2))
    186 self.conn.commit()

File "/opt/miniforge3/envs/latest-sotodlib/lib/python3.11/site-packages/sotodlib/core/metadata/obsfiledb.py:164, in ObsFileDb._get_version(self, conn)
    162 if conn is None:
    163     conn = self.conn
--> 164 rows = conn.execute('select value from meta where '
    165                     'param="obsfiledb_version"').fetchall()
    166 if len(rows) == 0:
    167     return None

OperationalError: no such column: "obsfiledb_version" - should this be a string literal in single-quotes?

```
-------------------------------------------------------

A similar issue also happened in:

 ```
 File "/opt/miniforge3/envs/latest-sotodlib/lib/python3.11/site-packages/sotodlib/core/context.py", line 270, in get_obs
    meta = self.get_meta(obs_id=obs_id, dets=dets, samples=samples,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/miniforge3/envs/latest-sotodlib/lib/python3.11/site-packages/sotodlib/core/context.py", line 466, in get_meta
    meta = self.loader.load(metadata_list, request, det_info=det_info, check=check,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/miniforge3/envs/latest-sotodlib/lib/python3.11/site-packages/sotodlib/core/metadata/loader.py", line 383, in load
    obs_info = self.obsdb.get(request['obs:obs_id'], add_prefix='obs:')
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/miniforge3/envs/latest-sotodlib/lib/python3.11/site-packages/sotodlib/core/metadata/obsdb.py", line 244, in get
    results = self.query(f'obs_id="{obs_id}"', add_prefix=add_prefix)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/miniforge3/envs/latest-sotodlib/lib/python3.11/site-packages/sotodlib/core/metadata/obsdb.py", line 317, in query
    c = self.conn.execute(q)
        ^^^^^^^^^^^^^^^^^^^^
sqlite3.OperationalError: no such column: "1755142326_280_GHz_Module" - should this be a string literal in single-quotes?
```

-------------------------------------------------------

I think an issue occurred in SQLite where `obs_id` and `obsfiledb_version` were treated as column names rather than text values. This happened because they were enclosed in double quotes ("...") instead of single quotes ('...'). As a result, the database returned an error when these identifiers didn’t match any column names. This leads to issues while trying to create ContextDB.

-------------------------------------------------------

I was not able to reproduce the issue in builds prior  to:
https://github.com/simonsobs/sotodlib/commit/89ed2e19d30c30b177457b010d1ef893122143f3

-------------------------------------------------------

**Possible Fix:**
Both these issues were fixed when I used single quotes for `obsfiledb_version` and `obs_id`, so that SQLite interprets them as string literals:

sotodlib/core/metadata/obsfiledb.py
https://github.com/simonsobs/sotodlib/blob/72660b5514fca740f98d254d2e7c61e8b124a559/sotodlib/core/metadata/obsfiledb.py#L164-L165
changed to:
```
        rows = conn.execute("select value from meta where "
                            "param='obsfiledb_version'").fetchall()
```

-------------------------------------------------------

sotodlib/core/metadata/obsdb.py
https://github.com/simonsobs/sotodlib/blob/72660b5514fca740f98d254d2e7c61e8b124a559/sotodlib/core/metadata/obsdb.py#L244
changed to:
`results = self.query(f"obs_id='{obs_id}'", add_prefix=add_prefix)`

iparask, 2025-02-21:
Hello @Ankurdev-astro at which resource did this error occur?

Ankurdev-astro, 2025-02-21:
Hello @iparask , I tested it on a stand-alone installation on a local machine (and also on a University Cluster):

To reproduce error, I made a new conda env:
`conda create -c conda-forge --name mlmap python=3.11.5 -y`
```
git clone https://github.com/simonsobs/sotodlib.git
cd sotodlib/
pip install . 
```

To reproduce error:

```
In [1]: import sotodlib

In [2]: import sqlite3, sys

In [3]: print(sotodlib.__version__)
0+untagged.3517.g72660b5

In [4]: print("SQLite Version:", sqlite3.sqlite_version)
SQLite Version: 3.49.1

In [5]: print("Python Version:", sys.version)
Python Version: 3.11.5 | packaged by conda-forge | (main, Aug 27 2023, 03:34:09) [GCC 12.3.0]

In [6]: from sotodlib.core import metadata

In [7]: metadata.ObsFileDb()
```

```
... /lib/python3.11/site-packages/sotodlib/core/metadata/obsfiledb.py:164, in ObsFileDb._get_version(self, conn)
    162 if conn is None:
    163     conn = self.conn
--> 164 rows = conn.execute('select value from meta where '
    165                     'param="obsfiledb_version"').fetchall()
    166 if len(rows) == 0:
    167     return None

OperationalError: no such column: "obsfiledb_version" - should this be a string literal in single-quotes?
```

mhasself, 2025-02-24:
Seems to be due to new sqlite3.  See [this](https://github.com/simonsobs/sotodlib/issues/1119), e.g.  Fixing this will require some careful revision of all our sqlite3-based code :(   For now try pinning sqlite3 to an earlier version?

PR #1126 (2025-02-28): sqlite3: remove double-quoted string literals
This is enforced starting at sqlite3.sqlite_version 3.49.1 or so.

Resolves #1119
closed 2025-02-28 by mhasself