simonsobs/sotodlib #1158: apply_pointing_model rotates azimuth by about 2pi

ykyohei, 2025-03-25:
I find that apply_pointing_model rotates azimuth by about 2pi, this would leads to problem in azimuth synchronous signal estimation/filtering because the azimuth goes to unexpected range. The amount of correction should be small.
 
```
aman = ctx.get_obs(obs_id = 'obs_1730092656_satp3_1011111',dets={'wafer.type': 'OPTC', 'wafer_slot': 'ws0', 'wafer.bandpass': 'f090', })
print(aman.boresight.az)
coords.pointing_model.apply_pointing_model(aman)
print(aman.boresight.az)
```

```
[3.63942698 3.63942719 3.63942706 ... 3.64619045 3.64614672 3.64610314]
[-2.66229314 -2.66229293 -2.66229307 ... -2.65552968 -2.65557341
 -2.65561699]
```

elle-shaw, 2025-03-27:
@mhasself,  Kyohei pointed out this issue to me about the pointing model corrections and I wonder if it is similar to something I have seen before when using quat.decompose_lonlat. For my purposes, I knew that the correct answer would never be a negative Az value, so I did a simple check like:

``
    if -1*negaz/DEG < 0:
        newaz = -1*negaz/DEG + 360
    else:
        newaz = -1*negaz/DEG

``
But I'll look into this more.

mhasself, 2025-03-27:
I'm interested in talking about this more... there are two options here:
1.  Adjust the "branch" of the output az to minimize the difference to the input az.
2.  Ask that people doing things like template binning use the `.ancil.az_enc` (or whatever it is called) -- i.e. the uncorrected, full range, encoder value.

I lean towards (2), and here's why:
- A minor problem with (1) is that in some cases the az perturbation won't be small.  For example, if you have base tilt, and are near zenith, then az can be basically anything.  (This is unlikely to affect our CMB obs, admittedly ... but I like it when I don't have to worry about such possible singularities.)
- An advantage of (2) is that the template results you derive are tied to _the data_, rather than to some kind of corrected data.  That means one is free to modify the pointing model, later, and the az-sync result can be re-used (it was based on az_enc, and thus can be applied based on that same az_enc). 

I'm open to (1), just raising a couple thoughts :)

ykyohei, 2025-03-28:
(2) is good point, thanks for pointing out
I will close this since https://github.com/simonsobs/sotodlib/pull/1162 is merged. Thank you for your quick fix.

closed 2025-03-28 by ykyohei