simonsobs/sotodlib #1161: Prevent NONE dets from entering obsfiledb

mhasself, 2025-03-27:
check_book has a "work-around" to suppress NONE dets, such as those that would appear due to use of fixed tones.  This work-around setting is on, in our update_obsdb jobs, however... site-pipeline `check_book` and `update_obsdb` both use a modified obsfiledb update call that dodges the NONE checking.

It's rare that these NONE happen but we should still guard against them properly.
- NONE dets must not be entered into the obsfiledb.
- the "work-around" should then, instead, be to ignore NONEs silently rather than to ignore them loudly.

And we need to eliminate the multiple obsfiledb update functions.

msilvafe, 2025-03-27:
We should be prepared for these NONE's to pop their head again. @tpsatt would like to start reading out fixed tones in the LAT to monitor readout phase drifts more regularly so we were planning to work together on updates on the sodetlib side to allow for setting these one.

mhasself, 2025-03-27:
Thanks for the headsup.

For that, if I could suggest _one thing_ on the bookbinding side (@kmharrington), it would be to note in the M_index.yaml (or whatever) _how many_ fixed tone / NONEs were detected (or something like that).  Then you can put that in the obsdb and find such acquisitions more easily.  Because we're not going to list those dets in the obsfiledb ... they'll be loaded as a separate entry in the TOD AxisManager.

mhasself, 2025-04-09:
Resolved by #1172

closed 2025-04-09 by mhasself