simonsobs/sotodlib #1190: Modify load_book to optionally allow loading in of NONE channels

msilvafe, 2025-04-17:
As discussed in issue #1161 we don't want to register NONE channels in the obsfiledb's or track them w/ the smurf-detsets however we do want to be able to load them if they do exist in the books. @tristpinsm has made an update to sodetlib https://github.com/simonsobs/sodetlib/pull/440 which myself and @tpsatt have tested and believe is ready to merge so we would like to start running with fixed tones on and loading them back at level 3. Currently we've tested that we can load in a frame w/ the spt3g software and see that there's a separate G3SuperTimestream field called untracked with the correct number and names of the fixed tones that we turned on in there. So really this should be a matter of: 1) @kmharrington adding the fact that there's untracked channels to the M_index.yaml in the bookbinder as suggested by @mhasself in #1161 and 2) modifying `load_book` to take an optional argument for weather or not to load the untracked channel information into a separate field (other than signal) in the axis manager.

Some books that we know have the untracked fields in them and can be used for testing are:
```
'/so/data/satp1/obs/17448/obs_1744840989_satp1_0000010/D_ufm_mv52r1_000.g3',
'/so/data/satp1/obs/17448/obs_1744845596_satp1_0000010/D_ufm_mv52r1_000.g3',
```

There's 23 untracked channels in those files with the following names:
```
[sch_NONE_0_038, sch_NONE_0_377, sch_NONE_2_326, sch_NONE_2_386, sch_NONE_3_029, sch_NONE_3_148, sch_NONE_3_375, sch_NONE_4_048, sch_NONE_4_070, sch_NONE_4_121, sch_NONE_4_152, sch_NONE_4_194, sch_NONE_4_204, sch_NONE_4_330, sch_NONE_5_107, sch_NONE_5_181, sch_NONE_5_275, sch_NONE_5_276, sch_NONE_5_349, sch_NONE_5_399, sch_NONE_5_455, sch_NONE_6_198, sch_NONE_6_362]
```

kmharrington, 2025-04-17:
Quick question where I haven't read this whole thing yet. For fixed tones, what about loading to something that isn't `signal`? They're separate in the books, what about loading to `aman.untracked` or `aman.tones` ?

msilvafe, 2025-04-17:
That is what I meant by 

> modifying load_book to take an optional argument for weather or not to load the untracked channel information into a separate field (other than signal) in the axis manager.

mhasself, 2025-04-18:
The book loader part is pretty straightforward, once there's a good example file to hack on -- thanks for that.

> we could even make a `tones` axis where we can put which bands and channels each of the fixed tones are on.

Is it presumptuous to calls these NONEs tones?  So how about a subaxisman with:

```
   tones (AxisMan)
       tdets  (LabelAxis)
       band[tdets]
       channel[tdets]
       signal[tdets,samps]
```
with band and channel extracted from the channel name?

I'm mostly asking about if these are the right names for things ... let me know.

msilvafe, 2025-04-18:
Those sound like sensible names for the sub axismanager and the axes to me.

kmharrington, 2025-05-12:
@mhasself  do you have files to hack on? There should be plenty from satp1 from around the time Max posted this issue.

kmharrington, 2025-05-12:
Also, could we look at this one and this issue: https://github.com/simonsobs/sotodlib/issues/1160 soon? Pairing them up because it's editing the same code.

mhasself, 2025-05-12:
Working on this ...

Since tone names aren't guaranteed unique between wafers, we need to either:
- reject the request to load the tones, in the case that several stream_ids are being concatenated and included in the output, OR
- rename tones, on the fly to include something unique, such as the stream_id, as a prefix.  (So `sch_NONE_0_377` -> `ufm_mv52r1_sch_NONE_0_377`)

kmharrington, 2025-05-12:
I vote rename tones

mhasself, 2025-05-12:
K, that's happening ... Actually can we just rename `sch_NONE_0_377` to `ufm_mv52r1_0_377` (and so on)?  We don't need the sch_NONE any more...

mhasself, 2025-07-11:
Addressed in #1219

closed 2025-07-11 by mhasself