simonsobs/sotodlib #1202: Different name cannot be allowed in class SubtractHWPSS

yoshinori-0778, 2025-04-30:
When I run `estimate_hwpss` and `subtract_hwpss` with different hwpss_stats_name `hwpss_stats_1`, I got an error because there is no proc_aman[self.hwpss_stats].
https://github.com/simonsobs/sotodlib/blob/2f329b4ba7f8480691befef865370f246366b333/sotodlib/preprocess/processes.py#L771-L772
A part of config is like below. This different name is needed for iterative glitch finding developed by Kyohei.
```
- name: "estimate_hwpss"
  calc:
      hwpss_stats_name: "hwpss_stats_1"
  save: True

- name: "subtract_hwpss"
    process:
      subtract_name: "hwpss_remove"
```

I temporary made a [branch](https://github.com/simonsobs/sotodlib/tree/hwpss_subtraction_bug) but this update has not backward compatibility. So I have not made a PR yet. This modification needs to set `hwpss_stats_name` in `subtract_hwpss` block explicity like this.
```
- name: "estimate_hwpss"
    calc:
      hwpss_stats_name: "hwpss_stats_1"
    save: True

- name: "subtract_hwpss"
    process:
      subtract_name: "hwpss_remove"
      hwpss_stats_name: "hwpss_stats_1"
```

We need to fix this when we incorporate the iterative glitch finding.

ykyohei, 2025-04-30:
I think this should work with no modification, we need 'hwpss_stats' in step_cfg not process_cfg. 
So, the issue is that we do not have example configuration in this process?
```
- name: "subtract_hwpss"
   hwpss_stats: "hwpss_stats_1"
   process:
        subtract_name: "hwpss_remove"
```

yoshinori-0778, 2025-05-01:
Thanks, yes, I confirmed it worked. Good to know.
I will make a PR for adding example configuration.

ykyohei, 2025-05-06:
Let me close as #1203 is merged

closed 2025-05-06 by ykyohei