simonsobs/sotodlib #1292: Question about subscan glitches IQR

mmccrackan, 2025-07-15:
When running some SATp2 observations through preprocessing, I saw some division by zero warnings when `subscan` was `True` for `get_glitch_flags`.  This was due to the last subscan being too short so the IQR was zero, which isn't unexpected.  However, I see that `iqr_range` which has length `aman.dets.count` is overwritten for each subscan here:

https://github.com/simonsobs/sotodlib/blob/73f839168b4eebc7badf419d81c7b3329e90b7dd/sotodlib/tod_ops/flags.py#L409C13-L409C22

When full_output is True, the full filtered signal is divided by this final subscan `iqr_range` (which might be all zeros as in the case described above) instead of the `iqr_range` of each subscan:

https://github.com/simonsobs/sotodlib/blob/73f839168b4eebc7badf419d81c7b3329e90b7dd/sotodlib/tod_ops/flags.py#L439

Is this intended or actually not a problem outside of this edge case (i.e. if the IQRs are not expected to vary significantly between scans)?

msilvafe, 2025-07-25:
> Yes this looks like a bug where the glitch_detection values in full_output will be a bit wrong if the iqr's vary significantly (which can happen). Flags should be okay

Ahh so you don't think this affects the flags? We were hoping this would address the issue that @jsyleung was seeing when trying to rerun glitches with the subscan flags basically he reported that all of the returned glitch flags were the same between w/ and w/o subscan detection which seemed to contradict your results/testing.

mmccrackan, 2025-08-11:
Okay, I will implement a fix that keeps a vector of IQRs and applies them to each of their respective subscans.

mmccrackan, 2025-10-31:
Closing since #1349 is merged.

closed 2025-10-31 by mmccrackan