simonsobs/sotodlib #1307: Tracking: Map Tracking Database

JBorrow, 2025-07-25:
This is a tracking issue for the Depth 1 and Map Tracking database, informed by the O1 session of the F2F.

For ensuring we can access D1 maps, we need a database that lists the location and creation time of those maps.

For the time domain pipeline, we need a database that connects D1 map entries to their processing status through the time domain pipeline.

For possible pointing offsets, we need a database that connects individual D1 maps to pointing offset measurements.

It would be helpful if...

- We maintained a list of all TODs that were used to create a D1 map
- A record of pipeline information (software and metadata versions) that were used to create a D1 map
- A 'sky coverage' patch for each D1 map (see below)
- We could track information about map quality and 'comments' from visual inspection
- This database was accessible at all computing centers

Map Information
---------------

We should have a table, `depth_one_maps` that contains the following information:

```
map_name -- Unique name (UUIDv7?) for map
map_path -- Non-localized path; use e.g. `DEPTH_ONE_MAPS_PATH=/so/data/maps:/so/data/depth1`
optics_tube
wafer
frequency
ctime
... -- Other map-specific metadata
```

Time Domain
-----------

We should have a table, `time_domain_processing` that contains the following information:

```
map_name -- Foreign key into `depth_one_maps`
processing_start
processing_status
processing_end
... -- Additional metadata for tracking, TBD
```

Pointing Offsets
----------------

We should have a table, `depth_one_pointing_offsets` that contains the following information:

```
map_name -- Foreign key into `depth_one_maps`
ra_offset
dec_offset
....
```

Time-Ordered Data
-----------------

We should have a table, `tod_depth_one` that contains the following information:

```
obs_id
map_name -- Foreign key into `depth_one_maps`
... -- Any TOD-specific metadata we need to remember
```

Pipeline Information
--------------------

We should have a table, `pipeline_information`, that contains the following information:

```
map_name -- Foreign key into `depth_one_maps`
sotodlib_version
map_maker
preprocess_info
...
```

Sky Coverage
------------

I propose that we split the sky into a 2N x N grid of 'sub-patches' such that we can label each D1 map with the patches that it covers. This will be an important optimization when we come back to the D1 maps to recreate light-curves. Each patch can be referred to by a unique identifier.

As such, we should have a table, `depth_one_sky_coverage`, that contains the following information.

```
patch_coverage -- Some integer or string for each patch
map_name -- Foreign key into `depth_one_maps`
...
```

JBorrow, 2025-07-25:
@axf295 @chervias I'd appreciate any thoughts on the above; we should start implementing alongside the depth one mapmaker.

JBorrow, 2025-07-25:
Here's what I'm thinking for the sky split up. We have a two byte integer: (RA * 54 / 360, Dec * 27 / 360) to uniquely define the squares.

[image]

axf295, 2025-07-26:
Some comments from the first look : 

**depth_one_maps table**

may want to make this general map table 

optics_tubes = ['ot1','ot2',...]
wafers = ['wafer1',...]
frequency='f090'
ctime_start=t1
ctime_end=t2
observations_included=[...]

**pointing_offsets**
Not sure how pointing offsets works.. might be model params or something but should that be attributed with the map itself? 
also think the time_domain_process should contain, in addition to start/end times and status,

a key of where it failed? i.e. matched filtering, forced photometry/pointing offset calc,etc.

Also from SPT it's good to have the ability to "permafail" observations which are known to be pathological such that the pipeline ignores it and any down-stream dependencies.

**TOD**

obs_ids or paths to TODs included in the depth1 map
map_name 

**sky_coverage**

I agree in theory with this. in practice I wonder how easy this will be. We will find out shortly I guess. 
The idea being given a source location, you could analyze only the depth1 maps which may overlap that position by searching his table?

Sulla2012, 2025-08-11:
Some comments:

>that contains the following:
> ...
> wafer
 

I thought we decided we were doing per-OT D1 maps as our most atomic? So no tracking wafer.

On pointing; the pointing model is a 9-param model detailed [here](https://github.com/simonsobs/sotodlib/blob/master/sotodlib/coords/pointing_model.py#L148) that is only updated when the LATR/mirrors/etc get moved. Hopefully this will not change ever but we may want to account for the possibility that it could. In addition, each obs has a two parameter correction, although that correction is computed from the maps so the logical flow is kind of the opposite of what's presented here. Finally, there's a per-det offset, which is the offset of each detector from the center of the focal plane. This again should only be updated if things (e.g., the receiver) move. All of these live on computing so the entry for the global pointing model could be a path to where that model lives.

 > ... -- Any TOD-specific metadata we need to remember

This metadata is already being tracked along with the TODs no? So is the point to save in the DB all the metadata one might want to make a selection on?

> patch_coverage -- Some integer or string for each patch

I mean I guess the dumbest thing is an integer/array of bool `1010001...` which maps onto the unrolled coverage array.

JBorrow, 2025-08-11:
From discussion:

a) Track the TOD-based pointing offset alongside the TODs that we track here that are used in each map.
b) Track a _pointing error_ for each D1 map (i.e. average RA, Dec offset and std deviation for point sources) as a function of time. Consider during D1 coaddition to use these tertiary pointing corrections.

chervias, 2025-08-29:
sorry I just had the chance to read this. One thing I don't understand is the preprocess table. Could you elaborate more? The details of the preprocessing do not need to be kept there, we will label the preprocessings and keep the detail elsewhere

Sulla2012, 2025-08-29:
We'd like to be able to sort on preprocessing details is the thought.

chervias, 2025-08-29:
Sorry, I'm afraid I still don't understand. Can you give an example? All depth-1 maps are preprocessed in the same way, what do you mean by "preprocessing details"?

JBorrow, 2025-09-04:
Not all depth-1 maps will remain on disk on site, so we'd like to track that site-pipeline version information. For instance, the version of sotodlib, python version, time that the pipeline was run and so forth. It's helpful to be able to select on these things in case we need to go back and fix maps. We would immediately be able to say 'we need A, B, C TODs to re-run the mapmaker because sotodlib v=9.1.2 was broken'